ekumenlabs/tangobot #93: Tangobot ROS.org wiki page does not refer to Play Store

juergensturm, 2017-05-11:
Currently none of the wiki pages on Tangobot mention that the app is available on Play Store. I would suggest to add it in several places to make sure that people are aware of it and can find it easily. Also, please link the video from the Tangobot main wiki page (similar to http://wiki.ros.org/tango_ros_streamer).

jubeira, 2017-05-11:
Yes, currently it's only mentioned in the installation step but there's no link to the app.
I will fix that!

jubeira, 2017-05-11:
Done!
@adamantivm I think the link in the Play Store is still pointing to the first version of the video.
I just put the link to the new one in Tangobot's main wiki page. Can you fix that?

adamantivm, 2017-05-11:
Yes, this is on my plate. Unfortunately it's not trivial and I haven't had the time to finalize it yet.
It will be done between today and early next week.

jubeira, 2017-05-16:
I think we can close this one; the video in the wiki and the playstore now point to the second version.

closed 2017-05-16 by jubeira

juergensturm, 2017-05-18:
Sounds good, thanks!