ekumenlabs/tangobot #103: Final tweaks for map_nav

adamantivm, 2017-06-05:
Any necessary tweaks for map_nav to work properly with Tangobot.

Off the bat:
 - Prevent 'set_pose' from changing the location on the map, since it doesn't do so on the robot

jubeira, 2017-08-22:
I think this one is done.
Perhaps (just thinking out loud here) it would be good to see how hard it is to subscribe to the inflation layer of the costmap, and display that layer with different colors. Right now, we are only displaying the occupancy grid, so the whole inflation layer is displayed as a black stain, which is not quite accurate.

jubeira, 2017-11-07:
Displaying different colors was not trivial, so we are sticking to black for now. In any case, that would be a separate issue.

closed 2017-11-07 by jubeira