ekumenlabs/tangobot #131: Timed out waiting for transform from /base_footprint to /map

JzHuai0108, 2023-06-29:
Dear Sir,

I am facing the below error and the global and local costmap do not show up when running move_base on an android phone.
```
2023-06-29 12:43:06.291 30657-30997 ros.costmap_2d          org.ollide.rosandroid                W  Timed out waiting for transform from /base_footprint to /map to become available before running costmap, tf error: . canTransform returned after 0.100423 timeout was 0.1.
```
For the android setup, the navigation rviz is given below, note the warnings about unavailable costmaps.
[image]

And the tf tree looks perfect as shown below.
[image]

Moreover, the move_base works perfectly on Ubuntu 18.

I tried to solve this issue with several approaches suggested elsewhere, but none worked.
The most promising one is [here](https://stackoverflow.com/questions/55406243/tf2-rosbuffercantransform-returning-false-for-existing-transformation). However, I wrote a test program comparing tf::TransformListener and tf2_ros::Buffer, both worked perfectly and could got the transform from /base_footprint to /map, on Ubuntu 18.
I currently have built the [roscpp_android](https://github.com/JzHuai0108/roscpp_android) with the kinetic ros libraries, but I mixed these libraries with the navigation modules of the melodic version, because I succeeded in testing these navigation modules on Ubuntu 18.
The roscpp_android built successfully,  the AMCL runs fine on android, and the move_base also works except that the costmap is unable to be created as discussed above.

Do I need to stick with kinetic tf::TransformListener in all navigation modules?

wfskxj, 2023-06-29:
However, when I run it on another device with Android 7.1, I encountered the following error：
org.ollide.rosandroid A/libc: /usr/local/google/buildbot/src/android/ndk-release-r18/external/libcxx/../../external/libcxxabi/src/abort_message.cpp:73: abort_message: assertion "terminating with uncaught exception of type pluginlib::LibraryLoadException: According to the loaded plugin descriptions the class costmap_2d::StaticLayer with base class type costmap_2d::Layer does not exist. Declared types are " failed

JzHuai0108, 2023-07-01:
I made sure that "LOCAL_WHOLE_STATIC_LIBRARIES" is added in the android.mk. I also made sure that the root failure cause is the failure to load the relevant plugin libraries. But the problem persists.

JzHuai0108, 2023-07-02:
Brute force makes miracles. I turned all runtime plugins into static libraries linked at compile time. Then, the move_base can show cost maps.

closed 2023-07-02 by JzHuai0108